dqrobotics/matlab #8: Renaming links() to n_links()

mmmarinho, 2019-03-28:
Hello, Bruno,

**Context** In the  `DQ_kinematics` object, there are two functions to retrieve the number of something
`links`
`n_dummy`

**Suggestion**
Change from  `links` to `n_link`

**Main reason**
It should be clear to users whenever the number/count of something is being returned. In that sense, having a convention such as `n_something` is a good practice.

What do you think?

Kind regards,
Murilo

bvadorno, 2019-03-28:
Hi, Murilo,

It makes sense. I'll change links to n_links. Just a side note: n_dummy will likely disappear in the near future because I intend to remove the so-called virtual joints (they usually cause unnecessary confusion and do not help much). What do you think?

Cheers,
Bruno

mmmarinho, 2019-03-29:
Hello, Bruno!

It might be a good choice to remove it in the future.

Regarding the function naming, I actually wrote the wrong name in the title but we can decide between
either 
`n_plural` or `n_singular`. 
So 
`n_link` and `n_dummy` or
`n_links` and `n_dummies`

Kind regards,
Murilo

bvadorno, 2019-03-29:
Hi, Murilo.

I usually prefer plural, but I think that both are fine. I think that the most important thing is to maintain the convention `**_n_**_something` to make it explicit that the variable refers to the number of something. This way, I'll maintain `n_links` and I won't change `n_dummy` for the moment as it is an internal variable that will be removed in the future.

Cheers,
Bruno

closed 2019-03-29 by bvadorno