dqrobotics/matlab #9: Method names in @DQ_cdts

mmmarinho, 2019-03-28:
Hello, Bruno,

**Context & Main Reason**
The method names in [DQ_cdts](https://github.com/dqrobotics/matlab/blob/master/%40DQ_cdts/DQ_cdts.m) are not compatible with the DQ robotics naming conventions.

**Suggestion**
Change
from `Jr()` to `relative_jacobian()`
from `Ja()` to `absolute_jacobian()`
from `jacobian1()` to `pose_jacobian_1()`    (see #7) 

Possibly changing `x1()`, `x2()`, `xa()`, and `xr()` to something else might also be useful. 
To keep the current convention
Instead of `x1()`, `fkm1()` or `fkm_1()` would be better. 

Don't want to over-complicate this discussion but we can change `DQ_kinematics::fkm()` to `DQ_kinematics::x()` or `DQ_kinematics::pose()`.

Kind regards,
Murilo

bvadorno, 2019-03-28:
Hi, Murilo.

I agree with `relative_jacobian()` and `absolute_jacobian()`, but I disagree with `DQ_kinematics::x()` because FKM is an established expression in robotics that has a very strong meaning. Concerning the cooperative variables, fkm1() would not make much sense as the cooperative variables do not necessarily depend on the forward kinematics, even if in this **particular** case `x1` is obtained by means of the forward kinematics. This way, I think that we could use absolute_pose, relative_pose, pose1, pose2.

Cheers,
Bruno

bvadorno, 2019-03-28:
I just realized that better than `relative_jacobian` and `absolute_jacobian` is `relative_pose_jacobian` and `absolute_pose_jacobian`.

bvadorno, 2019-03-28:
I did all the modifications to comply with our naming style, including the class name:
```
DQ_cdts -> DQ_CooperativeDualTaskSpace
xa -> absolute_pose
xr ->  relative_pose
Ja -> absolute_pose_jacobian
Jr ->  relative_pose_jacobian
x1 ->  pose1
x2 -> pose2
```

Cheers,
Bruno

closed 2019-03-28 by bvadorno