dqrobotics/matlab #11: Addition of plane(), line() , rotation(), and crossmatrix4()

mmmarinho, 2019-03-30:
**Context**
For the upcoming addition of the distance primitive Jacobians, some methods/functions are required.
I already added these functions to C++. We can still change their signature if there's something odd.

**Suggestion**
Addition of
1. `DQ::rotation()` and `DQ_robotics::rotation(DQ)`which returns the `P()` of a DQ but also checks if the DQ is unit norm. Non-unit-norm DQs have a valid `P() `but do not have a valid `rotation()`.
2. `DQ::crossmatrix4()` and `DQ_robotics::crossmatrix4(DQ)` which maps the left-side imaginary quaternion into a suitable matrix representing the cross-product so that a X b = crossmatrix4(a)*vec4(b)
3. `DQ_robotics::plane(n,pp)` which returns a plane with normal `n` and `pp` is a point in the plane
4. `DQ_robotics::line(l,pl)` which returns a line with direction `l` and `pl` is a point in the line

bvadorno, 2019-03-30:
Hi, Murilo. 

Suggestions 1 and 2 are fine, but I don't see a strict necessity of suggestions 3 and 4 because both plane and line are dual quaternions. Therefore, why do we need the class `DQ_robotics` and the corresponding `plane` and `line` methods?

Concerning the `rotation` method, what does it return in case of non-unit dual quaternions? Furthermore, what is the reason for having it in both the `DQ` and `DQ_robotics` classes? 

Cheers,
Bruno

mmmarinho, 2019-03-31:
Hello, Bruno!

I'm sorry for not being clear
By `DQ_robotics::something`, I just meant something in the `namespace DQ_robotics`. Not a new class. 

For example, most of the operators, e.g. as `P()`, we have as
method in `DQ`, such that `DQ::P()`
and as function in the `DQ_robotics` namespace 
`DQ_robotics::P(dq)`

Kind regards,
Murilo

bvadorno, 2019-04-01:
Hi, Murilo,

I think you missed part of my first question. =)

> 
(...) but I don't see a strict necessity of suggestions 3 and 4 because both plane and line are dual quaternions. Therefore, why do we need (...) the corresponding plane and line methods?

Cheers,
Bruno

mmmarinho, 2019-04-01:
Hello, Bruno

I'm guessing maybe the meaning somehow got lost due to the `DQ_robotics` confusion, but I meant
3. add to the `namespace DQ_robotics` the function `DQ plane(n,pp)` which returns a `DQ` representing the corresponding plane based on those two inputs.
```
DQ plane(const DQ& n, const DQ& ppi)
{
    if(n!=Im(n) || ppi!=Im(ppi))
    {
        throw std::range_error("One of the inputs is not imaginary in plane");
    }
    return (n+E_*(dot(ppi,n)));
}
```
4. add to the `namespace DQ_robotics` the function `DQ line(l,pl)` which returns a `DQ` representing the corresponding line based on the two inputs.
```
DQ line(const DQ& l, const DQ& pl)
{
    if(l!=Im(l) || pl!=Im(pl))
    {
        throw std::range_error("One of the inputs is not imaginary in line");
    }
    return (l+E_*cross(pl,l));
}
```

They are just auxiliary functions to generate planes and lines.

Kind regards,
Murilo

bvadorno, 2019-04-02:
Hi, Murilo,

I think that those convenience functions may backfire. For instance, a plane is given by `n + E_*d`, where `n` is normal to the plane and `d` is the signed distance from the plane to the origin of the reference frame. Given a point `p` in the plane, `d=dot(p,n)`, therefore `n + E_*d = n + E_*dot(n,p)` which is not an overhead compared to the convenience functions. 

One could argue that convenience functions would be helpful for people not well acquainted with dual quaternion algebra, but this is exactly the point where things may backfire. Consider that the person just needs to define a plane with normal given by 'n' located at a distance 'd' from the reference frame origin. The plane is already defined by these two variables, but the person may be tempted to find a point `p` in the plane and then use the function plane(n,p), which is extremely counterproductive and error-prone.

Cheers,
Bruno

mmmarinho, 2019-04-03:
Hello, Bruno,

I get your point.
In conclusion, let's aim only for the first two suggestions.

**Suggestion**
Addition of

```
DQ::rotation() and DQ_robotics::rotation(DQ)which returns the P() of a DQ but also checks if the DQ is unit norm. Non-unit-norm DQs have a valid P() but do not have a valid rotation().
    DQ::crossmatrix4() and DQ_robotics::crossmatrix4(DQ) which maps the left-side imaginary quaternion into a suitable matrix representing the cross-product so that cross(a,b) = crossmatrix4(a)*vec4(b)
```

Kind regards,
Murilo

bvadorno, 2019-04-14:
Done.

closed 2019-04-14 by bvadorno