dqrobotics/matlab #12: Addition of rotation_jacobian(), line_jacobian(), and plane_jacobian()

mmmarinho, 2019-03-30:
**Context**
As #11, this is relevant to the addition of the distance primitive Jacobians but also useful by themselves.
In the current implementation of these functions in C++, the functions in #11 are required.

**Suggestion**
Addition of:
1. `DQ_robotics::rotation_jacobian(pose_jacobian)` to retrieve the rotation Jacobian from the pose Jacobian
2. `DQ_robotics::line_jacobian(rotation_jacobian, translation_jacobian, eff_pose, line_direction_wrt_eff)` to retrieve the line Jacobian. The line_direction_wrt_eff makes this function very general, so the user can get the line Jacobian w.r.t the x-axis, y-axis, z-axis or any arbitrary axis
3. `DQ_robotics::plane_jacobian(rotation_jacobian, translation_jacobian, eff_pose, plane_normal_wrt_eff)` to retrieve the plane Jacobian with arbitrary normal w.r.t. the end-effector.

**Point for discussion**
There is a clear benefit of having such high-performance function signatures when implementing in C++, but they are verbose and might be a burden for users when performance is not a big requirement.

For instance, given
`DQ_kinematics robot = ` your favorite robot
`VectorXd           theta =` your robot's joint positions right now

to retrieve the `line_jacobian()` the following would be required:
`MatrixXd        J  = robot.pose_jacobian(theta)`
`DQ                 x  = robot.fkm(theta)`
`MatrixXd        Jt = translation_jacobian(J,x)`
`MatrixXd        Jr = rotation_jacobian(J)`
`MatrixXd       Jlz = line_jacobian(Jr,Jt,x,z_)`

So we should consider adding an alternative, possibly performance-unfriendly overloads of the Jacobian functions such that
`MatrixXd     Jlz  = line_jacobian(theta,z_)`

bvadorno, 2019-03-30:
Hi, Murilo.

Suggestion 1 is already implemented =). 

I think that even if we stick with the more verbose signatures, we can simplify them without having to sacrifice performance. For instance, consider 
```
MatrixXd J = robot.pose_jacobian(theta)
DQ x = robot.fkm(theta)
MatrixXd Jt = translation_jacobian(J,x)
MatrixXd Jr = rotation_jacobian(J)
MatrixXd Jlz = line_jacobian(Jr,Jt,x,z_)
```
Since `line_jacobian()` needs `Jr` and `Jt` and both `rotation_jacobian()` and `translation_jacobian()` need `J`, we can just do the following:
```
MatrixXd J = robot.pose_jacobian(theta)
DQ x = robot.fkm(theta)
MatrixXd Jlz = line_jacobian(J,x,z_)
```
This way, `Jt` and `Jr` are calculated inside `line_jacobian` and we use just three parameters. 

In any case, since Matlab does not support overloading functions using different signatures for the same function name (although it accepts a different number of parameters, which can be exploited to emulate different signatures), we must be careful in the way we choose the different signatures to ensure compatibility between C++ and Matlab.

Cheers,
Bruno

mmmarinho, 2019-03-31:
Hello, Bruno!

(Just to clarify as in #11, the `DQ_robotics::` means something in the `namespace DQ_robotics`.)

About the topic at hand, I guess that is the main point of discussion: to what extent do we want to balance optimization and verbosity (it this even a word?).

The most verbose, but most optimized
```
MatrixXd   J = robot.pose_jacobian(theta)
DQ         x = robot.fkm(theta)
MatrixXd  Jt = translation_jacobian(J,x)
MatrixXd  Jr = rotation_jacobian(J)
MatrixXd Jlz = line_jacobian(Jr,Jt,x,z_)
```

Somewhere in the center, as you suggested, in which we would have the overhead of computing `Jt` and `Jr` inside the function, even though that overhead is considerably smaller than computing `J` 
```
MatrixXd  J = robot.pose_jacobian(theta)
DQ        x = robot.fkm(theta)
MatrixXd Jlz = line_jacobian(J,x,z_)
```

And lastly the heaviest but least verbose
```
MatrixXd Jlz = line_jacobian(theta,z_)
```

Kind regards,
Murilo

bvadorno, 2019-04-01:
Hi, Murilo.

Thanks for the clarification about the namespace in issue #11. 

The intermediate solution that I suggested will be less performant than the most verbose one only in the case where we have multiple constraints that require Jt and Jr. Otherwise, in case of just one constraint, they will be equivalent because Jt will be calculated just once in both cases. Jr, on the other hand, will be just a copy of some elements, which usually is very fast. The heaviest one should be avoided, in my opinion.

Cheers,
Bruno

mmmarinho, 2019-04-01:
Hello, Bruno,

Then shall we settle with the signatures you suggested?
```
MatrixXd line_jacobian(pose_jacobian,pose,line_direction_wrt_eff)
MatrixXd plane(pose_jacobian,pose,pose,plane_normal_wrt_eff)
```

I ran some performance tests and the calculation of the `translation_jacobian()` and `rotation_jacobian()` were indeed not impactful when compared to `pose_jacobian.`

| Command | Time for 1 million iterations [ms] | Average time [ms] | 
| --- | --- | --- |
| `pose_jacobian()` | 27350.8 | 0.0273508 |
| `translation_jacobian()` | 512 |0.000512 |
|`rotation_jacobian()`|162.6|0.0001626|

Kind regards,
Murilo

bvadorno, 2019-04-01:
Hi, Murilo,

Yes, let us settle with those signatures.

Cheers,
Bruno

commit 91eab5cb (2019-04-19): Added the plane_jacobian() and a corresponding example. Closed #12.
closed 2019-04-19 by bvadorno