dqrobotics/matlab #16: Bug in plot function

juanjqo, 2019-04-08:
Hi @bvadorno 

There is a problem in plot function for DQ_kinematics objects in the last version. Sometimes, an undesired big black plane is plotted. 

I attached a minimal example. Here is my code:

> clear all
> close all
> clc
> 
> 
> larm_DH_a      = [0,     0,     0.011,      0.1572]; %0.05
> larm_DH_alpha  = [pi/2, -pi/2, pi/2,   pi/2];
> larm_DH_d      = [0.0285,    0.0182,    0.1482, 0 ];
> larm_DH_theta  =  [pi/2,  0, pi/2,   pi/2  ];
> 
> larm_DH_matrix = [larm_DH_theta;
>                         larm_DH_d;
>                         larm_DH_a;
>                         larm_DH_alpha]; 
> 
> larm  = DQ_kinematics(larm_DH_matrix,  'standard');
> 
> base =  - 0.70711 - 0.70711*DQ.j + DQ.E*( - 0.0014142 + 0.20771*DQ.i + 0.0014142*DQ.j - 0.26181*DQ.k);
> base = normalize(base);
> 
> theta_larm =  [  1.570796326794897
>                1.570796326794897
>                                0
>               -1.570796326794897];
>           
> larm.set_base(base);
> plot(larm, theta_larm);
> view([-57 29])
> axis equal

Also, I attached two images in order to show the problem. I changed the black color and I used the green color (for the undesired plane) to show that the robot is plotted.

[image]
[image]

[plot_minimal_example.txt](https://github.com/dqrobotics/matlab/files/3055586/plot_minimal_example.txt)

bvadorno, 2019-04-08:
Hi, Juancho.

I found the bug, but I'll push the fix in the next branch merge. For the moment, to prevent this problem you can just use a base frame such that the robot's first joint z-axis is not aligned with the world's x-axis.

Alternatively, if the former trick does not suit you, just change line 233 of DQ/plot.m from `v = DQ.j ` to `v = DQ.j*(plane_length/2); `

Cheers,
Bruno

juanjqo, 2019-04-08:
Ok, thanks!

bvadorno, 2019-04-08:
The fix is already available in the branch _mobile_base_.

juanjqo, 2019-04-08:
@bvadorno 
 The method 'set_base' was removed?  (I'm using the mobile_base branch)

bvadorno, 2019-04-08:
Yes for both questions. See https://github.com/dqrobotics/cpp/issues/9.

bvadorno, 2019-04-09:
@juanjqo, the legacy function `set_base()` has been pushed to the _mobile_base_ branch.

Cheers,
Bruno

juanjqo, 2019-04-09:
Great!

juanjqo, 2019-04-09:
Bruno, using the last version of the _mobile_base_ branch, there is a problem yet.
Now, an undesired black line is plotted.
I'm using the same minimal example and the same computer.
[image]

Cheers, 

Juancho

bvadorno, 2019-04-09:
Fixed in branch _mobile_base_.

closed 2019-04-09 by bvadorno