dqrobotics/matlab #18: Rename distance_jacobian?

mmmarinho, 2019-04-09:
**Context**

With the impending addition of several _pairwise_ distance Jacobians, namely

```
point_to_point_distance_jacobian()
point_to_line_distance_jacobian()
point_to_plane_distance_jacobian()
line_to_point_distance_jacobian()
line_to_line_distance_jacobian()
plane_to_point_distance_jacobian()
```

perhaps the current `DQ_kinematics.distance_jacobian(...)` method should be renamed for clarity.

**Suggestion**
Based on the _pairwise_ keyword I used in **Context**, perhaps a proper name would be
`point_distance_jacobian()`.

**Reasoning**
From a quick read of the implementation, the following seems to be true
`distance_jacobian(J,x)=point_to_point_distance_jacobian(J,x,0)`.
That is, the current
`distance_jacobian()` 
seems to be a special case of the
`point_to_point_distance_jacobian()` 
in which the workspace point is zero.

**Side effects**
Naturally we have to keep `distance_jacobian()` alive for a bit as a deprecated function, but we might also feel impelled to implement other _non-pairwise_ distance Jacobians if they make sense. 

**Discussion**
I only remember vaguely the paper in which the distance Jacobian was born, so I think @bvadorno  would be the best judge on whether we should rename this function or not.
With it actually being a special case, we might also consider removing it.

bvadorno, 2019-04-09:
Hi, @mmmarinho.

Indeed, the `distance_jacobian()` can be seen as a particular case of `point_to_point_distance_jacobian()`. Since it was first proposed in my 2010 paper, I couldn't foresee that we would add these many primitive Jacobians in the future. (Plus, the DQ Robotics wasn't born yet!) 

But there is a good reason, in my opinion, to keep this particular Jacobian, besides the fact that we must provide a legacy function. When we're working on more abstract spaces, such as the ones arising in cooperative manipulation, some important Jacobians arise, such as the `relative_pose_jacobian`. This Jacobian satisfies the relation` vec8(xr_dot) = Jr * q_dot`, where `xr_dot` is the time derivative of the relative pose between the two end-effectors and `q_dot = [q1_dot; q2_dot]` is the configuration space velocity of the cooperative system. Therefore, controlling the distance between the end-effectors is performed by using the `distance_jacobian` without having to worry about what is the reference point. Of course, the same could be achieved by using `point_to_point_distance_jacobian(Jr,xr,0)`, but this may have the side-effect of (maybe) misleading the more inexperienced user to think that 0 is the reference frame used by the absolute frame. Another point is that `distance_jacobian()` may be seen as a primitive closely related to `translation_jacobian()`.

Interestingly enough, even the relative_pose_jacobian could be renamed, if we wanted to, to `pose_to_pose_jacobian()`, because the `relative_pose_jacobian` relates two poses, and then the `pose_jacobian` would be just a particular case of `relative_pose_jacobian` (one of the poses would be the identity)! Summing up, even if the same result can be obtained in different ways and some Jacobians will be unavoidably particular cases of others, they are useful because they provide the right abstraction and expressivity in their particular contexts.

I hope I could make my point clear, as my response is quite long.

Cheers,
Bruno

mmmarinho, 2019-04-10:
Hello, @bvadorno 

All you said seems quite reasonable to me. Thanks for the explanation.

With the decision of keeping the method alive in mind, do you think we should rename it to `point_distance_jacobian()`? One plus side is that we give room to possibly other types of `xxx_distance_jacobian()` to be born in the future.

Kind regards,
Murilo

bvadorno, 2019-04-10:
Hi, @mmmarinho.

Maybe I'm quite biased with respect to this one, but I'd prefer to let the `distance_jacobian` untouched for the moment. In any case, even if we changed it to `point_distance_jacobian`, the legacy function name would have to exist for the sake of compatibility.

I suggest we wait till we do all other modifications/implementation before settling with the final decision. What do you think?

Cheers,
Bruno

mmmarinho, 2019-04-11:
Hello, @bvadorno,

Thanks for the reply.

Sure thing, we can wait and see. 

One thing we should really consider is to cut out all the deprecated functions before v1.0 since it will be the baptism of the library and the source code is starting to get cluttered.

Kind regards,
Murilo

bvadorno, 2019-04-12:
Hi, @mmmarinho.

It's a good idea to remove them just before v1.0. Are we going to adopt the year.month versioning system? If yes, then v1.0 would be 19.04, which defines a hard deadline. :grimacing:

Cheers,
Bruno

mmmarinho, 2019-04-14:
Hello, @bvadorno,

Yes, 19.04 is a good deadline for v1.0. Soon we have to stop with new additions and focus on squaring things out. 
Otherwise, v1.0 will never come. :satisfied:

Kind regards,
Murilo

bvadorno, 2019-04-14:
Hi, @mmmarinho, 🖖 

> Soon we have to stop with new additions and focus on squaring things out.
Otherwise, v1.0 will never come.

Yeah, young grasshopper, this is the side-effect of my current programming spree 👨‍🔬 🤓. Now that I finished the last example in serially coupled kinematic chains (have you seen the last two-legged example?), I'll finish the implementation of the most urgent issues, then we can proceed with the V-Rep example 🤖.

Cheers,
Bruno

P.S.: I'll close this issue as we already decided to put on hold the renaming decision.

closed 2019-04-14 by bvadorno