dqrobotics/matlab #20: Local variables in DQ_Kinematics

mmmarinho, 2019-04-10:
Hello, @bvadorno,

**Context**
I checked [`DQ_Kinematics`](https://github.com/dqrobotics/matlab/blob/mobile_base/robot_modeling/DQ_Kinematics.m) today and a few suggestions  came to mind.

**Suggestion**
For method arguments, perhaps naming them as `pose_jacobian` instead of `J` and `pose` instead of `x` would be helpful.

Moreover, some methods receive `theta` as argument but the robot configuration vector attribute is called `q`. Perhaps naming everything as `q` would be best.

Lastly, `ith` seems to provide little information about the argument's true purpose, so something like `to_link` or a similar name would be somewhat better.

**Reasoning**
I wouldn't say the "google" naming style isn't being followed, but consistency and intelligibility in the local variable naming is always welcomed by users and developers.

bvadorno, 2019-04-10:
Hi, @mmmarinho.

I totally agree that `q` is better than `theta` as the former is our standard notation for configuration. This is the effect of copy-and-paste from 'old' code. :grimacing:

I also agree that `ith` is not ideal inside a program, although this comes from the math notation "i-th link." But I don't think that `to_link` helps much in the context you mentioned, either. Maybe something like `to_ith_link`?

Finally, regarding `J` and `x`, since we have a standard convention inside DQ Robotics where `J` always refer to Jacobian matrices, I think that `J_pose` would suit the purpose of providing sufficient information without making any confusion with the function name `pose_jacobian`, besides being less verbose. On the other hand, `x` may be indeed very cryptic (and be mistaken for `x` coordinate) and can be improved. Since we also tend to use `x` (with an underbar) as the standard notation for pose in math calculations, maybe the slightly more verbose `x_pose` would solve the problem. What do you think?

Cheers,
Bruno

mmmarinho, 2019-04-11:
Hello, @bvadorno,

I don't know if you have the same functionality in MATLAB, but something I often resort to in C++ in my private code is to have highly-descriptive-but-verbose argument names and create a `const &` to them inside the functions to make the calculations easier to read and understand. Users need descriptive argument names, developers need easy-to-read calculations. There is AFAIK no performance impact.

E.g. 
```
MatrixXd DQ_Kinematics::distance_jacobian(const MatrxiXd& pose_jacobian, const DQ& pose)
{
  //Aliases
  const MatrixXd& J = pose_jacobian; //Alias for pose_jacobian, const & should be optimized out by the compiler
  const DQ&       x = pose;
  //Local variables
  const DQ t        = translation(x);
  const MatrixXd Jt = translation_jacobian(J,x);

  //Calculation
  return 2.0*vec4(t).transpose()*Jt; //Calculation is easy to read a similar to what we would write in a paper
}
```
(This is obviously an overkill for such a simple function so it is not how it is implemented in C++ now, just an example to clarify my point)

If this functionality exists and is reasonably simple to implement in MATLAB, we could use
`pose` and map it inside the function to `x`,
`j_pose` or `pose_jacobian` and map it to `J` or `Jx`
and so on.

Otherwise we can stick to what you suggested
| old | new |
| --- | --- |
| `theta` | `q` |
| `J` | `J_pose` |
| `x` | `pose_x` |

What do you think?

Kind regards,
Murilo

bvadorno, 2019-04-11:
Hi, @mmmarinho.

I agree with you. We should avoid cryptic names as best as possible. Unfortunately, Matlab does not accept the definition of reference variables, therefore any alias would make a copy of the function argument, which unavoidably has a performance impact, at least in terms of memory usage. Let us settle with what I suggested?

Cheers,
Bruno

mmmarinho, 2019-04-12:
Hello, @bvadorno,

Let's go with your suggestion. :)

| current argument name | new argument name |
| --- | --- |
| `ith` | `to_ith_link` |
| `J` | `J_pose` |
| `x` | `pose_x` |

Kind regards,
Murilo

closed 2019-04-12 by bvadorno