dqrobotics/matlab #21: Toolbox for MATLAB

mmmarinho, 2019-04-11:
**Context**

I've been checking [Peter Corke's Robotics Toolbox](https://petercorke.com/wordpress/toolboxes/robotics-toolbox) recently and I noticed he also packs his library as a [MATLAB toolbox](https://petercorke.com/wordpress/toolboxes/robotics-toolbox). 

**Suggestion**
Pack the major versions of dqrobotics-matlab into [MATLAB toolboxes](https://petercorke.com/wordpress/toolboxes/robotics-toolbox).

**Reasoning**
Since we went the extra mile by making a PPA and automatic uploads to PyPi, making a toolbox for major versions would also look pro.

bvadorno, 2019-04-11:
Hi, @mmmarinho.

Great idea! After our major development sprint, I'll take care of it.

Best regards,
Bruno

bvadorno, 2019-10-06:
Done and tested for Matlab R2014b on MacOS. I updated the installation instructions on our webpage accordingly.

Cheers,
Bruno

closed 2019-10-06 by bvadorno