dqrobotics/matlab #22: Old website link in Matlab files.

juanjqo, 2019-04-11:
Hi @bvadorno,

The new DQ robotics website is not updated in the Matlab files.
Current:
% DQ Robotics website: dqrobotics.sourceforge.net

Expected:
% DQ Robotics website: https://dqrobotics.github.io/

Cheers,

Juancho

bvadorno, 2019-04-19:
Hi, Juancho.

I'll be fixing the weblinks in the documentation whenever I encounter an old one, but since we have tons of files now, I may skip some of them (and since this issue has low priority, I won't have time to look in each one of the files). Therefore, if you find some file that needs fixing, please add them as an item in a task list (see https://guides.github.com/features/mastering-markdown/). You can do it directly in this issue, by editing your text and I'll tick them whenever I fix the file. Thanks.

Cheers,
Bruno

juanjqo, 2019-04-19:
Ok, I'll do it.  =)

bvadorno, 2019-04-21:
Great. I'll close this issue now, but we can reopen it in the future in case you find more of those "legacy" files.

Cheers,
Bruno

closed 2019-04-21 by bvadorno