dqrobotics/matlab #23: Attribute dim_configuration_space in DQ_MobileBase should be an attribute of parent class DQ_Kinematics

mmmarinho, 2019-04-12:
Hello, @bvadorno,

**Context**
I'm checking the implementation of `DQ_MobileBase.m` and came across the following attribute

(in file [`DQ_MobileBase`](https://github.com/dqrobotics/matlab/blob/master/robot_modeling/DQ_MobileBase.m))
```
properties (Access = protected)
        dim_configuration_space;
end
```
I don't know if I'm misunderstanding MATLAB, but I found that strange since the method related to that attribute is part of DQ_Kinematics

(in file [`DQ_Kinematics`](https://github.com/dqrobotics/matlab/blob/master/robot_modeling/DQ_Kinematics.m))
```
% GET_DIM_CONFIGURATION_SPACE returns the dimension of the configuration
% space.
dim = get_dim_configuration_space(obj);
```

**Suggestion**

I think the attribute `dim_configuration_space` should also be part of `DQ_Kinematics`, since the abstract method `get_dim_configuration_space(obj)` is part of `DQ_Kinematics`.

Is there anything I'm missing?

Kind regards,
Murilo

bvadorno, 2019-04-12:
Hi, @mmmarinho,

> I think the attribute dim_configuration_space should also be part of `DQ_Kinematics`, since the abstract method `get_dim_configuration_space(obj)` is part of `DQ_Kinematics`.

Indeed, it could be from a programming point of view. The main reason for this difference is that we were using `n_links` in the old `DQ_kinematics`, which now is called `DQ_SerialManipulator`, to store the dimension of the configuration space. This makes sense in that context because the number of links equals the number of joints, which equals the dimension of the configuration space in **serial manipulators**. 

In most mobile robots, however, there are no links (except maybe in trailler robots), therefore `n_links` is not an appropriate name. The more uniform solution, which would be to use **only** `dim_configuration_space` in the parent class `DQ_Kinematics` would not be appropriate in case of serial manipulators, because `n_links` makes much more sense throughout the whole class. 

The alternative solution, which consists in having both `n_links` and `dim_configuration_space` is redundant and, in my opinion, not necessary; after all, the method `get_dim_configuration_space(obj)` can be used to retrieve the right information, regardless of the variable used to store it in a specific class. In case of heterogeneous kinematic chains, this issue becomes still more evident as the topology of the configuration space is composed of several different spaces, and each one of them can be related to different variables, and the function `get_dim_configuration_space` is used to add their dimensions to provide the final dimension of the configuration space.

> In fact, there seems to be some mistake because of the following method in which base_frame() returns base_pose and not base_frame.
> Do they have a different functionality? Is this duplication expected?

They had in an early version, but not anymore. Thank you for pointing that out. We can safely remove the attribute `base_pose` and the method `base_frame(obj)` in `DQ_MobileBase`. The `fkm()` method should be updated accordingly.

> Is there a reason why raw_fkm() and raw_pose_jacobian() are abstract methods of DQ_MobileBase() instead of being abstract methods of DQ_Kinematics()?

Yes. In fact, every robot must have the methods `fkm()` and `pose_jacobian`, but not all of them must necessarily have their `raw` counterparts. Incidentally, for all robots we _currently_ have, the `raw` methods are useful, but not necessarily for future robots. This way, this is just a 'future-proof' implementation for classes that inherit from DQ_Kinematics to prevent needless implementation of dummy `raw` methods in subclasses that don't need them.

Cheers,
Bruno

mmmarinho, 2019-04-12:
Hello, @bvadorno,

Thanks for the reply:

> The alternative solution, which consists in having both `n_links` and `dim_configuration_space` is redundant and, in my opinion, not necessary; after all, the method [...]

I get your point, I don't think the existence of `n_links` is a problem if `n_links` isn't exposed outside of the class as it is now. 
If we have both `get_dim_configuration_space()` and `get_n_links()` to expose the same thing, I guess as long as it makes the code easier to understand I'll back it up. 
My question stemmed from the fact that currently in MATLAB, some attributes that require access control are publicly exposed, which means that the user, in MATLAB, does not need to use the accessor functions  to get and set these attributes. 
This means that they can circumvent the accessors we give them and that can lead to confusion for us and users in the future. For example, in MATLAB, the following can be done
```
links = ax18.n_links
links = ax18.get_dim_configuration_space()
```
which might be fine but this is also possible:
```
ax18         = Ax18ManipulatorRobot.kinematics();
ax18.n_links = 1; 
%Your code is now broken
```
and
```
ax18.set_reference_frame(reference_frame) ; %This checks if the DQ is unit
ax18.reference_frame = reference_frame; %This doesn't check if the DQ is unit
```
which is strange.
Although my first comment was somehow misguided towards `n_links`, now that I checked the code a little more deeply I think it is important to consider revising the access control in MATLAB and provide accessors when necessary, because I'll replicate them as close as possible in C++ and Python for compatibility between languages.
(Although the C++ overlords of old will ask the developer to add accessors for everything because it's future-proof, forcing all attributes to have accessors is something [not recommended](https://github.com/google/styleguide/blob/gh-pages/pyguide.md) and not particularly necessary, so we can stick to what is really needed.)
I can open a new issue about this if you think it's relevant.
 
> Yes. In fact, every robot must have the methods `fkm()` and `pose_jacobian`, but not all of them must necessarily have their `raw` counterparts. Incidentally, for all robots we _currently_ have, the [...]

Abstract parent classes are indeed very poisonous so it is good to be careful. We can retake this discussion when the `DQ_Controller` classes are re-written.

Kind regards,
Murilo

bvadorno, 2019-04-12:
Hi, @mmmarinho.

As a matter of fact, `n_links` can (and should) be a protected attribute because its use is internal only. Furthermore, since future classes such as `DQ_ParallelManipulator` will probably inherit from `DQ_SerialManipulator`, it is convenient to make it protected instead of private. In any case, we will prevent those issues you mentioned. Let's settle with a protected `n_links` and only the  `get_dim_configuration_space` method? (`reference_frame` can be protected as well.)

Cheers,
Bruno

closed 2019-04-14 by bvadorno