dqrobotics/matlab #30: Wrong Jacobian using DQ_Kinematics.line_to_line_distance_jacobian [BUG]

juanjqo, 2019-04-29:
**Bug description**

 Hi @bvadorno!

The method DQ_Kinematics.line_to_line_distance_jacobian() is not working properly. The same code in Python returns a different Jacobian. I implemented the function and my results are the same to the Python version.
I check the Matlab code and I think the possible bugs are:
 
Line 176 from DQ_Kinematics.m
Current:
`-0.5*(hamiplus8(workspace_line) +...`
Expected:
`-0.5*(haminus8(workspace_line) +...`

Line 177 from DQ_Kinematics.m
Current:
`haminus8(workspace_line))*line_jacobian`
Expected:
`hamiplus8(workspace_line))*line_jacobian`

Line 201 from DQ_Kinematics.m
Current:
`if ~mod(phi,pi)`
Expected:
`if mod(phi,pi)`

I performed that changes and the results match with the Python version and my local implementation.

*Code*
Minimal example in Matlab 
```
clear all
close all
clc

robot = KukaLwr4Robot.kinematics;
q = [0,0.5,0,0,0,0.5,0];
x = robot.fkm(q);
J = robot.pose_jacobian(q);

% DQ plucker line         

r = x.P;
t = x.translation;
lz = r*DQ.k*r';
mz = cross(t, lz);
DQ_lz = lz + DQ.E*mz;  % define DQ Dynamic line

% DQ static line
l = DQ([0,0,0,1]);
tl = DQ([0,1,1,0]);
m = cross(tl,l);
DQ_l =  l + DQ.E*m;

Jlz2   = DQ_Kinematics.line_jacobian(J, x, DQ.k);
Jlz_l2 = DQ_Kinematics.line_to_line_distance_jacobian(Jlz2,  DQ_lz, DQ_l)
```
*Output*
```
Jlz_l2 =

  Columns 1 through 5

   0.724989770311299   0.200380136942568   0.287071347746572  -0.270095207844456   0.287071347746572

  Columns 6 through 7

   0.338067401973798  -0.000000000000000
```

**Expected behavior**
- I expect the same results obtained in Python/C++.
This is, by performing the changes that I mentioned:

```
Jlz_l2 =

  Columns 1 through 5

  -2.000000000000000  -0.000000000000001  -1.571072894301144                   0  -1.571072894301144

  Columns 6 through 7

  -0.000000000000000  -0.000000000000000
```

The same minimal example in Python:

```
from dqrobotics import *
from dqrobotics.robots         import KukaLw4Robot
from dqrobotics.robot_modeling import DQ_Kinematics
from dqrobotics.robot_modeling import DQ_CooperativeDualTaskSpace
from numpy import hstack, vstack
import numpy as np

q = np.array([0,0.5,0,0,0,0.5,0])

robot = KukaLw4Robot.kinematics()
x = robot.fkm(q)
J = robot.pose_jacobian(q)

# DQ plucker line

r = x.P()
t = translation(x)
lz = r*DQ.k*conj(r)
mz = cross(t, lz)
DQ_lz = lz + DQ.E*mz  #define DQ Dynamic line

# DQ static line
l = DQ([0,0,0,1])
tl = DQ([0,1,1,0])
m = cross(tl,l)
DQ_l =  l + DQ.E*m

#np.set_printoptions(precision=3)
Jlz2   = DQ_Kinematics.line_jacobian(J, x, DQ.k)
Jlz_l2 = DQ_Kinematics.line_to_line_distance_jacobian(Jlz2,  DQ_lz, DQ_l)
print(Jlz_l2)
```

*Expected output*
```
[[-2.00000000e+00  0.00000000e+00 -1.57107289e+00  0.00000000e+00
  -1.57107289e+00  0.00000000e+00 -3.33066907e-16]]
```

**Environment:**
 - OS: [Ubuntu 18.04]
 - dqrobotics version  0.1.1.21 (For Python)
- I did the git pull operation for Matlab today (April 29, 2019)
 - Python version [3.6.7]
 - MATLAB version [R2016b]

Cheers, 

Juancho.

bvadorno, 2019-05-01:
Hi, @juanjqo.
Indeed, there is a bug, but only in line 201.
Current:
`if ~mod(phi,pi)`
Expected:
`if mod(phi,pi)`

Note that line 176 and 177 implements a single operation, namely
`
Jinner = -0.5*(hamiplus8(workspace_line) + haminus8(workspace_line))*line_jacobian;`
which is equivalent to what you suggested
`Jinner = -0.5*(haminus8(workspace_line) + hamiplus8(workspace_line))*line_jacobian;
`
Therefore, the problem was happening due to bug in line 201, which made the algorithm calculate the Jacobian for parallel lines when lines were non-parallel and vice versa.

Pro tips:
1 - If you use `include_namespace_dq`, the aliases `i_, j_, k_,` and `E_` become available for you;
2 - The adjoint transformations `Ad()` and `Adsharp()` are already implemented in all three languages. Therefore, you can just use `lz = Ad(r,k_)` instead of `lz = r*DQ.k*r'`. 

Also, this code 
```
l = DQ([0,0,0,1])
tl = DQ([0,1,1,0])
m = cross(tl,l)
DQ_l =  l + DQ.E*m
```
can be elegantly written as
`line = k_ + E_*cross( i_+ j_ , k_)`
or, if you want a more verbose version as you did before:
```
l = k_,
tl = i_ + j_;
m = cross(tl,l);
line = l + E_*m;
```
Note that `DQ_l` is not an adequate name, since it does not follow neither the Google Python style nor DQ Robotics convention. As a matter of fact, `DQ_Something` is usually reserved for standard DQ Robotics classes.

Cheers,
Bruno

Cheers,
Bruno

closed 2019-05-01 by bvadorno

juanjqo, 2019-05-01:
@bvadorno 
Thanks for the tips.