dqrobotics/matlab #31: FKM of the DQ_WholeBody does not return intermediate transformations of one of the chains [BUG]

StUrb, 2019-04-30:
**Bug description**
Hello, @bvadorno,

I am currently working with the DQ_WholeBody  serializing two legs in order to perform a walking algorithm. One of my steps is to calculate the center of mass of the serialized chain. My approach for such calculation starts with the intermediate transformations of each link of the serialized chain. However, when I use functionality fmk(q,joint) it does not work properly.

**To Reproduce**
I used the code in the two_legs_example to create a minimal example.

*Code*

```
    % Include the DQ namespace to use i_ instead of DQ.i, etc.
    include_namespace_dq
    
    % The legs are composed of two KUKA LWR4 robots coupled together    
    right_leg = KukaLwr4Robot.kinematics();
    right_leg.name = 'Right leg';
    left_leg = KukaLwr4Robot.kinematics();
    left_leg.name = 'Left leg';
    
   
    % The left foot will be located 20 cm to the left, along the x axis,
    % but with the z-axis pointing downwards.
    left_foot = (1 - E_ * 0.5 * 0.2 * i_) * i_;
    % The right foot will be located 20 cm to the right, along the x axis,
    % but with the z-axis pointing downwards.
    right_foot = (1 + E_ * 0.5 * 0.2 * i_) * i_;
    
    
    % Physically place the left leg in the workspace
    x_0_to_left_effector = left_leg.fkm(zeros(7,1));
    new_base_frame = left_foot .* T(x_0_to_left_effector);
    left_leg.set_base_frame(new_base_frame);
    % Physically place the right leg in the workspace
    x_0_to_right_effector = right_leg.fkm(zeros(7,1));
    new_base_frame = right_foot .* T(x_0_to_right_effector);
    right_leg.set_base_frame(new_base_frame);
    
    %% Initializes the robot serially coupled kinematic chain from the mobile base
    left_to_right = DQ_WholeBody(left_leg, 'reversed');
    % The base frame is the left foot. The reference frame, also located at
    % the left foot, provides the transformation with respect to the global
    % frame. Therefore, any element in the chain will be given with respect
    % to the global frame.
    left_to_right.set_reference_frame(left_foot);
    left_to_right.set_base_frame(left_foot);
    
    % Determines the rigid transformation between the base frames of the
    % two legs
    base_left_to_base_right = left_leg.base_frame'*right_leg.base_frame;
    % Add this constant transformation to the chain
    left_to_right.add(base_left_to_base_right);    
    % Finally, add the right frame
    left_to_right.add(right_leg);
    
    q = zeros(14,1);
    x1 = left_to_right.fkm(q,1)
    x2 = left_to_right.fkm(q,2)
    x3 = left_to_right.fkm(q,3)
    x4 = left_to_right.fkm(q,4)
    x5 = left_to_right.fkm(q,5)
    x6 = left_to_right.fkm(q,6)
```
*Output*
```
x1 = 
         (1i) + E*(0.1 + 0.55j)
x2 = 
         (1i) + E*( - 0.1 + 0.55j)
x3 = 
         (1i) + E*( - 0.1)
Index exceeds matrix dimensions.

Error in DQ_WholeBody/raw_fkm (line 170)
                if isa(obj.chain{i}, 'DQ_Kinematics')

Error in DQ_WholeBody/fkm (line 142)
                x = obj.reference_frame * raw_fkm(obj,q,ith);

Error in issue_FKM_minimal_example (line 52)
    x4 = left_to_right.fkm(q,4)
```

**Expected behavior**
- I was expecting the frames of each joint of the first chain (left leg in the minimal example) 

*Expected output*
This is the output of the left leg frames without serialization.
```
x1 = 
         (0.70711 + 0.70711i) + E*(0.1096j + 0.1096k)
x2 = 
         (1) + E*(0.155k)
x3 = 
         (0.70711 - 0.70711i) + E*( - 0.25102j + 0.25102k)
x4 = 
         (1) + E*(0.355k)
x5 = 
         (0.70711 + 0.70711i) + E*(0.38891j + 0.38891k)
x6 = 
         (1) + E*(0.55k)
```

**Environment:**
 - OS: [Ubuntu 16.04]
 - dqrobotics 28/04/2019
 - MATLAB version [R2016b]

bvadorno, 2019-05-01:
Hi, @StUrb. 

Thank you for the suggestion, I'll work on it now as it was already on my task list. 😉 

This "problem" you mentioned is rather a limitation than a bug. Notice that, according to the documentation, 
> `x = fkm(q, ith)` calculates the forward kinematics up to the `ith`  kinematic chain.

Since you have only three kinematic chains (two legs and one constant transformation) in the serialized chain, `ith` can never be greater than 3. 

Since the behavior matches exactly what the function was designed for, it is not a bug (by definition). It is, of course, one important limitation that must be overcome to solve problems like yours. I'll post the solution in a few minutes. Stay tuned!

Cheers,
Bruno

StUrb, 2019-05-01:
Hello @bvadorno,

Thank you for your answer!  :)

Cherrs,

Stella

bvadorno, 2019-05-01:
Hi @StUrb, done. Please let me know if you encounter any issue. 

@mmmarinho, could you please implement those changes in C++? I'll move this issue to the C++ repository.

Cheers,
Bruno