dqrobotics/matlab #33: The method line_to_line_squared_distance() returns a wrong distance [BUG]

juanjqo, 2019-05-07:
**Bug description**
- The method DQ_Geometry.line_to_line_squared_distance(line1, line2) returns a wrong distance.
In this minimal example I show the problem. I define two lines that are perpendicular. Furthermore, the distance between them is 1.0. But the method returns zero.

*Code*

```
clear all 
close all
clc

include_namespace_dq

l1 = k_ + E_*cross(0 , k_);
l2 = i_ + E_*cross(j_ , i_);

plot(DQ(1), 'scale', 0.1);
hold on
plot(l1,'line', 1);
axis equal
plot(l2,'line', 1);

angle = acos(P(dot(l1,l2)))
distance = DQ_Geometry.line_to_line_squared_distance(l1,l2)
```
*Output*
```
angle =

   1.570796326794897

distance = 
         0
```
[image]

**Expected behavior**
- I expect that the right distance. In this case, d=1.

*Expected output*
```
distance = 
         1
```

**Environment:**
 - OS: [e.g. Ubuntu 16.04]
 - dqrobotics version (I updated the dqrobotics for Matlab today, May 7th, 2019)
 - MATLAB version [2016b] 

**Additional context**
- I think that the problem is on the line 105 of DQ_Geometry.m

Current:
` if P(l1_dot_l2) ~= 0`
Expected:
` if P(l1_dot_l2) == 0`

bvadorno, 2019-05-10:
Hi, @juanjqo.

Thank you, the bug has been fixed. Line 105 was indeed incorrect, as I was checking if the lines are **_not_ perpendicular**, but the correct would be to check if they are **_not_ parallel**. Your suggestion, however, wouldn't work either, because it would check if the lines are **perpendicular**. (Your suggestion works with your example, incidentally, because the lines are perpendicular...) Therefore, the appropriate check is `P(l1_cross_l2) ~= 0`. Alternatively, one equivalent statement would be `if P(l1_dot_l2) ~= 1`, assuming that both lines have unit norm; however, someone might want to use lines with different norms in the future, so I decided to use the cross product instead.

Cheers,
Bruno

closed 2019-05-10 by bvadorno

juanjqo, 2019-05-10:
@bvadorno 
Thanks!