dqrobotics/matlab #45: [REQUEST] VrepInterface does not comply with Google Python's style guide

bvadorno, 2019-08-17:
The Google Python's style guide states that the maximum line length for Python code is [80 characters](https://google.github.io/styleguide/pyguide.html#32-line-length).

Since the Matlab implementation uses the Google Python's style, all lines should be constrained to 80 characters. 

I already fixed the `set_object_translation()` method, but other methods still have way too long lines, which is annoying when using small screens such as the ones of Macbook Pro. 😥 

**Suggestion**: Fix all remaining methods to comply with Google Python's style.

mmmarinho, 2019-08-17:
Will do!

bvadorno, 2020-04-17:
Hi, @mmmarinho,

Could you please fix this issue for our upcoming version? Thank you!

Cheers,
Bruno

commit b4bbd94a (2020-04-20): Fix #45
closed 2020-04-23 by mmmarinho