dqrobotics/matlab #50: [QUESTION] Support for dummy joints?

mmmarinho, 2019-10-08:
Hello, @bvadorno,

I remember we discussed about dropping support for dummy joints in `DQ_SerialManipulator`, but it seems that they are still alive in release 19.10.0.

Are we planning on removing them soon?

bvadorno, 2019-10-13:
Hello, @mmmarinho. 

Yes, we are. As soon as we finish our tutorial paper, we can think about those things. Also, I think it's time to build a roadmap to guide our next steps. What do you think?

Cheers,
Bruno

mmmarinho, 2020-02-13:
Hello, @bvadorno,

I think that's a good idea.
(We talked about this before in a meeting but forgot to reply on github)

Cheers,
Murilo

bvadorno, 2020-04-17:
Hello, @mmmarinho, 

I removed the support for dummy joints in my [development branch](https://github.com/dqrobotics/matlab/tree/updating_to_20_04_bva). Since I'll be doing a pull request soon, I'll close this issue.

Cheers,
Bruno

closed 2020-04-17 by bvadorno