dqrobotics/matlab #51: [BUG] MATLAB interface with V-REP

frederico-afonso, 2019-10-10:
**Bug description**
When using the MATLAB interface with V-REP, the following functions used to communicate with the robot,
   - get_joint_positions();
   - set_joint_target_positions();
   - set_joint_positions();
   - stop_simulation();

do not work properly unless a command "get_joint_positions()" is used previously to any attempt to call any of them. 

**To Reproduce**
Without uncommenting  the line "vi.get_joint_positions(joint_names);" the following code will not be able to move the robot on V-REP. Additionally, the simulation will not stop.

*Code*

```
joint_names = {'Jaco_joint1','Jaco_joint2','Jaco_joint3','Jaco_joint4','Jaco_joint5','Jaco_joint6'};

vi = VrepInterface();
vi.connect('127.0.0.1',19997);
vi.start_simulation();
pause(1)
% vi.get_joint_positions(joint_names);

q = [0.0; 2.9; 1.3; 4.2; 1.4; 0.0];
vi.set_joint_target_positions(joint_names, q');
pause(10)

vi.stop_simulation();
vi.disconnect();
```

**Expected behavior**
- The robot should move to the position given by "q". This will only happen if the line "vi.get_joint_positions(joint_names);" is uncommented.

**Environment:**
 - OS: Ubuntu 18.04
 - dqrobotics version: v19.10.0
 - MATLAB version: R2017b
 - V-REP version: PRO EDU V3.6.2. (rev. 0)

**Additional context**
- Joint names were given considering the use of the JACO robot.

mmmarinho, 2019-10-11:
Hello, @frederico-afonso 

Thanks for the report.

I created the pull request #52 to fix this issue. While it's not merged with the master branch you can use the `issue51` branch.

Note that `VrepInterface` will be renamed to `DQ_VrepInterface`.

After the proposed fix, the following code works fine on my machine:
```
joint_names = {'Jaco_joint1','Jaco_joint2','Jaco_joint3','Jaco_joint4','Jaco_joint5','Jaco_joint6'};

vi = DQ_VrepInterface();
vi.connect('127.0.0.1',19997);
vi.start_simulation();

q = [0.0; 2.9; 1.3; 4.2; 1.4; 0.0];
vi.set_joint_target_positions(joint_names, q);
pause(1);

vi.stop_simulation();
vi.disconnect();
```

frederico-afonso, 2019-10-11:
Thank you, @mmmarinho ! 

This really solved the problem. I will be using `issue51` branch for now.

PR #52 (2019-10-13): Renames VrepInterface to DQ_VrepInterface and fixes issue #51
Two fixes for `DQ_VrepInterface`:
1. Renames `VrepInterface` and `VrepInterfaceMapElement` to have the `DQ_` prefix, as we discussed previously.
2. Fixes #51. The bug stemmed from a `pausecommunication` command that I added with good intentions but turns out wasn't a good idea 👎 .
closed 2019-10-13 by bvadorno