dqrobotics/matlab #53: [QUESTION] A few suggestions for the changes in commit dc115096

mmmarinho, 2020-01-23:
Hello, @bvadorno,

I was checking the changes in 
https://github.com/dqrobotics/matlab/commit/dc11509645ecc0516187e00e1bad2699f0113ca8
to translate them into cpp and so on.

I have a few suggestions.
- [ ]  The local variable `stability_counter_max` has a setter method called `set_max_stability_counter`. It would probably be best to unify the naming here
- [ ] On https://github.com/dqrobotics/matlab/blob/dc11509645ecc0516187e00e1bad2699f0113ca8/robot_control/DQ_KinematicController.m#L188 instead of `if controller.target_primitive`, maybe it would be best to check if it is a plane with `is_plane(controller.target_primitive)`
- [ ] The same for https://github.com/dqrobotics/matlab/blob/dc11509645ecc0516187e00e1bad2699f0113ca8/robot_control/DQ_KinematicController.m#L254
- [ ] On https://github.com/dqrobotics/matlab/blob/dc11509645ecc0516187e00e1bad2699f0113ca8/robot_control/DQ_KinematicController.m#L368 we should probably set `is_stable_=false` before or after the counter is set to zero. Otherwise, the flag won't be changed back `false` when the system leaves the stable region.

Please let me know if I misunderstood something!

Cheers,
Murilo

bvadorno, 2020-01-30:
Hi @mmmarinho ,

Thank you for the feedback. You got it right. I have changed the code following your suggestions and I'll push it to the server soon. I will change the name of the method is_stable, however, to address a remark made by one of our paper's reviewers.

Cheers,
Bruno

closed 2020-01-30 by bvadorno

mmmarinho, 2020-02-07:
Hello, @bvadorno, 

I forgot one suggestion which might also be important in a few situations. I will not reopen the issue because I'll leave you to judge if this comment is relevant now or not.

In line 366, the stability counter is increased whenever the task error is below a threshold.
https://github.com/dqrobotics/matlab/blob/6aad5ee37a69881f3857d2e0a3a643b882e8adc7/robot_control/DQ_KinematicController.m#L366
Before incrementing the `stability_counter`, we might want to check if it is below the maximum allowed value. Like so
```matlab
if controller.stability_counter < controller.stability_counter_max
    controller.stability_counter = controller.stability_counter + 1; 
end
```

The reason for this is to prevent an overflow if the code keeps running for a very long time when the robot already reached a stable region.

You might ask me: "but is that even possible?". In regular situations, no. For a robot operating in a 10ms control loop, it would take around 248 days of continuous operation to overflow (reach the maximum value of 2147483647 for a 32-bit signed int). In simulations, that problem might be apparent much faster.

Cheers,
Murilo

bvadorno, 2020-02-09:
Hi, @mmmarinho.

> The reason for this is to prevent an overflow if the code keeps running for a very long time when the robot already reached a stable region.

Yes, I agree.

> You might ask me: "but is that even possible?".

Actually, I might not, for the simple fact that not preventing a possible, even if improbable, overflow is simply bad coding. 😉  Thank you for spotting it.

I've implemented your suggestion.

Cheers,
Bruno

commit 6aad5ee3 (2020-01-30): Close #53