dqrobotics/matlab #62: The pose_jacobian method of DQ_holonomicBase class returns a wrong Jacobian[BUG]

juanjqo, 2020-08-01:
**Bug description**

Hi @bvadorno !

- The pose_jacobian method of DQ_holonomicBase class returns a wrong Jacobian. Specifically, the problem is the line [73](https://github.com/dqrobotics/matlab/blob/e99b9df3a4764fa6d6015467bcec929557d3c24e/robot_modeling/DQ_HolonomicBase.m#L110) of the DQ_HolonomicBase Class. The correct is: j73 = 0.25*(-x*c - y*s); instead of j73 = 0.25*(x*c - y*s); 

*Code*

```
clear all
close all
clc

q = [0.0003,   -0.0003,    0.0158];
x =  q(1);
y =  q(2);
phi = q(3);

%------Implemented version of the Jacobian-------%
j13 = -0.5*sin(phi/2);
j62 = -j13;
j71 = j13;

j43 = 0.5*cos(phi/2);
j61 = j43;
j72 = j43;

j63 = (1/4)*(-x*sin(phi/2) + y*cos(phi/2) );
j73 = (1/4)*(-x*cos(phi/2) - y*sin(phi/2) );

Jhol = [0 0 j13
        0 0  0
        0 0  0
        0 0 j43
        0 0 0
        j61 j62 j63
        j71 j72 j73
        0 0 0];
%--------------------------------------

holonomic_base = DQ_HolonomicBase();
J = holonomic_base.pose_jacobian(q, 2)
Jhol
error = J-Jhol
```
*Output*
```
J =

         0         0   -0.0039
         0         0         0
         0         0         0
         0         0    0.5000
         0         0         0
    0.5000    0.0039   -0.0001
   -0.0039    0.5000    0.0001
         0         0         0

Jhol =

         0         0   -0.0039
         0         0         0
         0         0         0
         0         0    0.5000
         0         0         0
    0.5000    0.0039   -0.0001
   -0.0039    0.5000   -0.0001
         0         0         0

error =

   1.0e-03 *

         0         0         0
         0         0         0
         0         0         0
         0         0         0
         0         0         0
         0         0         0
         0         0    0.1500
         0         0         0
```

**Expected behavior**
- I expect the same result of the implementation of Jhol in equation 4.11 of your [Phd](https://adorno.eng.ufmg.br/publications/phd_thesis_final_version.pdf) dissertation.

*Expected output*
```
J =

         0         0   -0.0039
         0         0         0
         0         0         0
         0         0    0.5000
         0         0         0
    0.5000    0.0039   -0.0001
   -0.0039    0.5000   -0.0001
         0         0         0

Jhol =

         0         0   -0.0039
         0         0         0
         0         0         0
         0         0    0.5000
         0         0         0
    0.5000    0.0039   -0.0001
   -0.0039    0.5000   -0.0001
         0         0         0

error =

         0         0         0
         0         0         0
         0         0         0
         0         0         0
         0         0         0
         0         0         0
         0         0         0
         0         0         0
```

**Environment:**
 - OS: Ubuntu 18.04
 - dqrobotics version (e99b9df)

 - MATLAB version  2018a

Cheers!

bvadorno, 2020-08-02:
Hello, @juanjqo,

Thanks for catching this bug. It has been fixed.

Cheers,
Bruno

closed 2020-08-02 by bvadorno

juanjqo, 2020-08-02:
Thanks @bvadorno!  The bug is present in Python and C++ as well. I opened a new issue.