dqrobotics/matlab #65: Issue when using "doc DQ" in Matlab

maribernardes, 2020-09-16:
When trying to open the toolbox documentation in Matlab, I get the error message:

Exception in thread "AWT-EventQueue-0": java.lang.NullPointerException
	at com.mathworks.help.helpui.DocRoot.buildDocPageUrl(DocRoot.java:44)
	at com.mathworks.help.helpui.DocPageUrlBuilder.buildDocPageUrl(DocPageUrlBuilder.java:25)
	at com.mathworks.help.helpui.DocPageUrlBuilder.buildFileUrl(DocPageUrlBuilder.java:21)
	at com.mathworks.mlwidgets.help.LightweightHelpPanel$JsHelpPanelClientRunnable.getUrlToLoad(LightweightHelpPanel.java:224)
	at com.mathworks.mlwidgets.help.LightweightHelpPanel$JsHelpPanelClientRunnable.run(LightweightHelpPanel.java:208)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

The "help DQ" is working fine.
Just starting a new course and hoping to recommend this toolbox for students' assignments ;D

bvadorno, 2020-09-16:
Thanks for the feedback, @maribernardes.

Could you please provide more details? What is the operating system you are using? Which Matlab version? 

When you refer to "toolbox documentation," do you mean the one you are supposed to access when you type the command `doc DQ`, for example? 

Thank you,
Bruno

bvadorno, 2020-10-16:
Hi, @maribernardes,

Since this issue doesn't seem to be happening with other users and you didn't give me enough feedback to further investigate it, I'll close it for now.

If anything new comes up, please let me know and I'll reopen the issue.

Cheers,
Bruno

closed 2020-10-16 by bvadorno