dqrobotics/matlab #73: [QUESTION] Does the DQ_ClassicQPController supports the DQ_FreeFlyingRobot?

marcos-pereira, 2022-07-05:
Dear DQ Robotics developers,

As in the title:  Does the DQ_ClassicQPController supports the DQ_FreeFlyingRobot? 
If yes, I tried to use it as below and some errors appeared. If not, then that may be something for the next versions.

To check this, I have tried to run the following piece of code (sorry in advance for some redundancy in the code):

```
include_namespace_dq;

% Initial pose
p_init = 0*i_ + 0*j_ + 0*k_;
phi_init = 0;
n_init = 1*i_ + 0*j_ + 0*k_;
r_init = cos(phi_init/2.0) + n_init*sin(phi_init/2.0);
x_init = r_init + E_*0.5*p_init*r_init;
pose_references{1} = x_init;

% Test Pose 1
p1 = 60*i_ + 60*j_ + 20*k_;
phi1 = 0;
n1 = 1*i_ + 0*j_ + 0*k_;
r1 = cos(phi1/2.0) + n1*sin(phi1/2.0);
x1 = r1 + E_*0.5*p1*r1;
pose_references{2} = x1;

poses_trajectory = poses_references

robot = DQ_FreeFlyingRobot();
robot.set_radius(0.5);

sampling_time = 0.01;    
desired_error = 0.009;

% Controller
gain = 1;
dq_solver = DQ_QuadprogSolver;
controller = DQ_ClassicQPController(robot, dq_solver);
controller.set_gain(gain);
controller.set_stability_threshold(desired_error);
controller.set_control_objective(ControlObjective.Pose);

% Initial pose 
init_pose = pose_references{1};

% Current task pose
task_pose = init_pose;
task_variable = vec8(task_pose);

figure

% Plot task pose references
for pose = 1:length(pose_references)
    plot(pose_references{pose}, 'scale', 6);
    hold on
end

task_pose_reference = poses_trajectory{2};

while ~controller.system_reached_stable_region()

    %compute the control input u    
    uvec = controller.compute_setpoint_control_signal(task_variable, vec8(task_pose_reference));            
    
    task_variable = vec8(task_pose);
    task_variable = task_variable + sampling_time*uvec;
    task_pose = DQ(task_variable);
    
    plot(robot,task_pose);            
    drawnow;    
end
```

This code results in the following error:
```
Undefined function 'vec8' for input arguments of type 'double'.

Error in DQ_KinematicController/get_task_variable (line 275)
                    task_variable = vec8(x_pose);

Error in
DQ_TaskspaceQuadraticProgrammingController/compute_setpoint_control_signal
(line 58)
                task_variable = controller.get_task_variable(q);

Error in InterpolatePoseTrajectory (line 70)
            uvec = controller.compute_setpoint_control_signal(task_variable,
            vec8(task_pose_reference));
```

Based on the error there seems to be some inconsistency when choosing the `task_variable`, i.e. the configuration vector `q`, for the controller. Based on the [DQ_FreeFlyingRobot.m](https://github.com/dqrobotics/matlab/blob/master/robot_modeling/DQ_FreeFlyingRobot.m) comments, the free-flying robot configuration vector is its pose. Because of that I have tried using `task_variable = vec8(task_pose)`. If we follow the error in the `compute_setpoint_control_signal`, we arrive at the method `get_task_variable` in the [DQ_KinematicController.m](https://github.com/dqrobotics/matlab/blob/master/robot_control/DQ_KinematicController.m). The method will use the `task_variable=vec8(task_pose)` to calculate the `x_pose`, which is obtained through the robot FKM resulting in `x_pose=x=task_variable` (see line 72 from [DQ_FreeFlyingRobot.m](https://github.com/dqrobotics/matlab/blob/master/robot_modeling/DQ_FreeFlyingRobot.m)), to then get back the `task_variable=vec8(x_pose)`. 

Some possible questions to be investigated:
1) Before going deeper into the code, there is the title question: does the controller supports the free-flying robot in its own conception? 
2) If yes, then why is the DQ_FreeFlyingRobot fkm() returning a "double"? 
Note that if we add the following lines right before `uvec=...` in the code above
```
....
robot_fkm_pose = robot.fkm(task_variable)
controller.get_task_variable(robot_fkm_pose)
uvec = controller.compute_setpoint_control_signal(task_variable, vec8(task_pose_reference));            
....
```
we get the error as expected
```
Undefined function 'vec8' for input arguments of type 'double'.

Error in DQ_KinematicController/get_task_variable (line 275)
                    task_variable = vec8(x_pose);

Error in InterpolatePoseTrajectory (line 70)
            controller.get_task_variable(robot_fkm_pose)

```
3) Was the choice `task_variable=vec8(task_pose)` correct in the code above? Is the configuration vector of the free-flying robot the `task_pose` as it is?

Best regards,
Marcos

bvadorno, 2022-07-13:
Hi @mmmarinho,

You are exactly right. One way of using `DQ_ClassicQPController` with `DQ_FreeFlyingRobot` is to use make a suitable projection for the Jacobian matrix in a way that one can "map" the left-invariant error to the Euclidean error, which will then be compatible with `DQ_ClassicQPController`, as in the paper 

> H. L. Pham, B. V. Adorno, V. Perdereau, and P. Fraisse, “Set-point control of robot end-effector pose using dual quaternion feedback,” Robotics and Computer-Integrated Manufacturing, vol. 52, pp. 100–110, Aug. 2018, DOI: 10.1016/j.rcim.2017.11.003.

Best regards,
Bruno

closed 2022-07-13 by bvadorno

mmmarinho, 2022-07-14:
@bvadorno  Thanks!

@marcos-pereira In conclusion, `DQ_ClassicQPController` does not currently support `DQ_FreeFlyingRobot`. 
As suggested by @bvadorno , for now you can inherit from `DQ_FreeFlyingRobot` and overload the Jacobian accordingly (if needed).

marcos-pereira, 2022-07-29:
Hello guys,

I have tested using the projection as in the paper: 

> H. L. Pham, B. V. Adorno, V. Perdereau, and P. Fraisse, “Set-point control of robot end-effector pose using dual quaternion feedback,” Robotics and Computer-Integrated Manufacturing, vol. 52, pp. 100–110, Aug. 2018, DOI: 10.1016/j.rcim.2017.11.003.

I created a class DQ_FreeFlyingRObotConstrained that inherits from the DQ_FreeFlyingRobot:
```
classdef DQ_FreeFlyingRobotConstrained < DQ_FreeFlyingRobot
        
    methods (Access = public)

        function N = pose_jacobian(~,x,~)
        % POSE_JACOBIAN(x) returns, given the unit dual quaternion x that
        % represents the free-flying robot configuration, the constrained pose Jacobian
        % N 
            if nargin > 3
                error('Invalid number of arguments');
            else
                % calculates the mobile robot pose Jacobian. If an index is
                % passed as an argument, just ignore it because there is
                % only one element in the chain. However, we have to tackle
                % this case to ensure compatibility with other
                % DQ_Kinematics objects.

                J = haminus8(x/2);
                D8 = diag([1.0 -1.0 -1.0 -1.0 1.0 -1.0 -1.0 -1.0]);
                N = -haminus8(x)*D8*J;
            end
        end         

    end
end
```

The difference from the projection used in the mentioned paper is the negative sign that was added because the DQ_ClassicQPController considers  task_error=task_variable-task_reference and in the paper, they used the opposite.

I wrote the following code that runs the constrained motion controller on the modified free-flying robot. I add spherical restricted region constraints. The robot must go to `x1` without colliding with sphere1.

```
clc 
close all
clear all

include_namespace_dq;

robot_radius = 0.5;

% OBSTACLES
spheres_centerpoints = {};
spheres_radius = {};

sphere1_center = 30*i_ + 30*j_ + 10*k_;
sphere1_centerpoint = vec3(sphere1_center);
sphere1_radius = 5;
spheres_centerpoints{1} = sphere1_centerpoint;
spheres_radius{1} = sphere1_radius;

safe_distance_spheres = robot_radius;

% Initial pose
p_init = 0*i_ + 0*j_ + 0*k_;
phi_init = 0;
n_init = 1*i_ + 0*j_ + 0*k_;
r_init = cos(phi_init/2.0) + n_init*sin(phi_init/2.0);
x_init = r_init + E_*0.5*p_init*r_init;

% Test Pose 1
p1 = 60*i_ + 60*j_ + 20*k_;
phi1 = 0;
[... 107 lines cut ...]
Error in DQ_FreeFlyingRobot/update_robot (line 191)
            p = vec3(translation(q));

Error in DQ_FreeFlyingRobot/plot (line 85)
            update_robot(robot,q);

Error in issue1_freeflyingrobot (line 121)
    plot(robot,task_pose);
```

Note that I print the `task_pose_norm` and it is, as expected, not 1.

# Outcome without problem in the task pose norm
I also tested the controller without setting the inequality constraints and it worked correctly. To test this case just comment the line that set the inequality constraints
```
% controller.set_inequality_constraint(constraint_matrix,...
    %     constraint_vector);
```
The robot goes to the pose `x1`.

# Possible issues
1) Maybe I have used the wrong projection for the Jacobian matrix. If the issue is the projection for the Jacobian matrix, how should I begin to obtain it? Any hints on the starting point?
2) I am considering the system a single-integrator system. I tested using the numerical integration of unit dual quaternions, but it also results in non-unit dual quaternions.
3) I also tested some very low control gains. The norm approaches 1, but the control signal becomes very small. I also tested to vary the damping factor and get similar outcomes.
4) Maybe there is some typo in my constraint definition? I double-checked but is worth checking it again.

marcos-pereira, 2022-08-02:
Hello guys,

I have found one solution to the problem I mentioned above.

My projection for the Jacobian matrix was not precisely correct. I investigated the following paper

> Savino, Heitor & Pimenta, Luciano & Shah, Julie & Adorno, Bruno. (2019). Pose consensus based on dual quaternion algebra with application to decentralized formation control of mobile manipulators. Journal of the Franklin Institute. 357. 10.1016/j.jfranklin.2019.09.045. 

and adapted the constrained controller to obtain a controller in the logarithmic space (logspace) using the Q8 matrix and the free-flying robot Jacobian. This way, I did not need to change anything in the DQ_FreeFlyingRobot.m. As a result, the logspace controller worked with and without the constraints activated.

Just let me know if you need more details about the implementation.

Best regards,
Marcos