dqrobotics/matlab #74: [LANGUAGE INCOMPATIBILITY] The DQ_KinematicController does not reset the stable region flag.

marcos-pereira, 2022-08-02:
**Describe the missing/unexpected functionality**
The [DQ_KinematicController.m](https://github.com/dqrobotics/matlab/blob/master/robot_control/DQ_KinematicController.m) reset_stability_counter() does not reset the `system_reached_stable_region_` as in the C++ version [DQ_KinematicController.cpp](https://github.com/dqrobotics/cpp/blob/master/src/robot_control/DQ_KinematicController.cpp).

Personally, I have used this functionality in C++ and missed it in Matlab. However, I am not sure if there should be a separate method to reset the system_reached_stable_region or not.

As a temporary solution, I just added the functionality myself.

If needed, I could do a pull request to add this functionality two weeks from now. 

**Environment:**
 - dqrobotics version: the version from the Master branch.

bvadorno, 2023-04-02:
Hi @marcos-pereira,

If you want to continue this issue, please adapt it to our current standards. The information you gave me needs to be more comprehensive. More specifically:

**Describe the missing/unexpected functionality**
- A clear and concise description of what is happening.
- Add error messages to the output (if any)
- MAKE A MINIMAL EXAMPLE TO SHOW WHAT YOU WANT TO SHOW; THE LONGER YOUR CODE IS, THE LESS LIKELY YOU ARE TO GET PROPER SUPPORT

**Matlab behavior** (if applicable)

*MATLAB SCRIPT*

```matlab
%% Your MATLAB script here
```

*MATLAB OUTPUT*

```bash
[Paste your output here]
```

Many thanks,
Bruno

ffasilva, 2024-08-17:
Hi @marcos-pereira,

Thank you for the minimal example. This issue was solved in #111.

@bvadorno, could you please mark this as solved?

Kind regards,
Frederico

closed 2024-08-18 by bvadorno