dqrobotics/matlab #76: [BUG]

juanjqo, 2022-12-21:
**************************
### Acceptance of the Code of Conduct
_Leave these in your bug report, read, and check them._ 
- [x] I understand that support for DQRobotics is given voluntarily. 
- [x] In understand that any possible response and its timeliness will be based on the relevance, accuracy, and politeness of my request and the following discussion.
- [x] If a DQRobotics member replies, I will let them know, politely, if their response solves my issue or not. 
*************************

**Bug description**
- At the time being, the classes `DQ_SerialManipulatorDH` and `DQ_SerialManipulatorMDH` only support revolute and prismatic joints. However, both class constructors allow any type of joint of the class `DQ_JointType`. 

**To Reproduce**

For example, this code creates a serial manipulator using types of joints different from revolute and prismatic ones. 

*Code*

```
clear all;
clc;
close all;

ax18_DH_theta   = [0       0      -pi/2   -pi/2      -pi/2];  % theta
ax18_DH_d       = [0.167   0       0     0.1225          0];  % d
ax18_DH_a       = [0      0.159    0.02225    0      0.170];  % a
ax18_DH_alpha   = [-pi/2   0      -pi/2   -pi/2          0];  % alpha

H  = double(DQ_JointType.HELICAL);
S  = double(DQ_JointType.SPHERICAL);
C  = double(DQ_JointType.CYLINDRICAL);
PL = double(DQ_JointType.PLANAR);
SD = double(DQ_JointType.SIX_DOF);

ax18_DH_type    = [SD PL H S C];

ax18_DH_matrix = [  ax18_DH_theta;
    ax18_DH_d;
    ax18_DH_a;
    ax18_DH_alpha;
    ax18_DH_type;
    ]; 

ax = DQ_SerialManipulatorDH(ax18_DH_matrix)
```
*Output*
```
ax = 

  DQ_SerialManipulatorDH with properties:

               theta: [0 0 -1.5708e+00 -1.5708e+00 -1.5708e+00]
                   d: [1.6700e-01 0 0 1.2250e-01 0]
                   a: [0 1.5900e-01 2.2250e-02 0 1.7000e-01]
               alpha: [-1.5708e+00 0 -1.5708e+00 -1.5708e+00 0]
                type: [6 2 7 3 4]
    JOINT_ROTATIONAL: 1
     JOINT_PRISMATIC: 2
            effector: [1×1 DQ]
             plotopt: {}
             lineopt: {'Color'  'black'  'Linewidth'  [2]}
              handle: []
             n_links: 5
                name: '0.049469'
     reference_frame: [1×1 DQ]
          base_frame: [1×1 DQ]
                   q: []
```

**Expected behavior**
- I expect an exception pointing out the use of unsupported types of joints.

*Expected output*

Something like:
```
Error using DQ_SerialManipulator/check_type_of_joints
Unsupported type of joints. Valid type of joints: DQ_JointType.REVOLUTE,
DQ_JointType.PRISMATIC.
```

**Environment:**
 - OS: [e.g. Windows 11]
 - dqrobotics version (current version of the master branch)
 - MATLAB 2022a]

bvadorno, 2022-12-21:
Hi @juanjqo, 

May you implement it in a new pull request? There is no rush because I’ll be off for one week. 

Best wishes,
Bruno

juanjqo, 2022-12-23:
Hi @bvadorno, sure!

PR #84 (2023-02-02): This commit fixes the bug #76
@dqrobotics/developers 

Hi @bvadorno,

This PR proposes the following modifications to fix #76 :

- Removed the public property `type`  from subclasses `DQ_SerialManipulatorDH` and `DQ_SerialManipulatorMDH`. Added the protected property `joint_types` in `DQ_SerialManipulator`.  Motivations:
  - The joint types do not depend on any specific parametrization. Therefore, `obj.joint_types` is a property common to all subclasses, like `obj.n_joints`.
  - Since we now have the method `set_joint_types` (and get_joint_types, both added in this PR) there is no need to keep [... cut]