dqrobotics/matlab #77: [BUG] Overloading DISPLAY is not recommended

htadashi, 2022-12-24:
**************************
### Acceptance of the Code of Conduct
_Leave these in your bug report, read, and check them._ 
- [x] I understand that support for DQRobotics is given voluntarily. 
- [x] In understand that any possible response and its timeliness will be based on the relevance, accuracy, and politeness of my request and the following discussion.
- [x] If a DQRobotics member replies, I will let them know, politely, if their response solves my issue or not. 
*************************

**Bug description**
Not exactly a bug, but mostly a code QA suggestion. MATLAB's code analyzer recommends overloading `disp` instead of `display` because the built-in `display` function already calls `disp` and the behavior between `disp` and `display` may differ. 

More details are available at: https://de.mathworks.com/help/matlab/matlab_oop/displaying-objects-in-the-command-window.html

**To Reproduce**
Using `display(DQ(1))` shows
```
" =  
        1 "
``` 
instead of the expected output.

**Expected behavior**
It should print the Expected output below.

*Expected output*
```
" 1 "
```

**Environment:**
 - OS: Windows 11
 - dqrobotics version 20.04.0.1
 - MATLAB version R2022b

**Additional context**
Merry X-mas! 🎄

bvadorno, 2023-02-02:
@htadashi, thank you for pointing that out. May you make the pull request again? I had to rebase the code, which messed up several pull requests.

Thanks!

htadashi, 2023-02-02:
@bvadorno Sure! :)
I opened a new draft PR in #86.

bvadorno, 2023-04-01:
@htadashi, I finally had time to look at it. All good. I accepted your pull request.

Many thanks,
Bruno

closed 2023-04-01 by bvadorno