dqrobotics/matlab #79: [QUESTION] Feature request: add code auto-completion

htadashi, 2023-01-01:
### Acceptance of the Code of Conduct
_Leave these in your question, read, and check them._ 
- [x] I understand that support for DQRobotics is given voluntarily. 
- [x] In understand that any possible response and its timeliness will be based on the relevance, accuracy, and politeness of my request and the following discussion.
- [x] If a DQRobotics member replies, I will let them know, politely, if their response solves my issue or not. 

### Question

Since MATLAB R2021b it is possible to customize its editor's [code suggestion and completion](https://www.mathworks.com/help/releases/R2021b/matlab/matlab_env/check-syntax-as-you-type.html#bswj2of-1) by adding a json file in the folder that contain the class code. More details can be seen [here](https://www.mathworks.com/help/matlab/matlab_prog/customize-code-suggestions-and-completions.html).
As MATLAB is not a strongly-typed language, it would be nice to have this feature for DQ robotics classes.

Thank you for considering this request.

bvadorno, 2023-04-01:
Hi @htadashi,

I think it's an excellent suggestion but we don´t have enough bandwidth to implement it. Would you like to lead this initiative and implement it?

Many thanks,
Bruno

htadashi, 2023-04-01:
Hi @bvadorno, I am glad that you liked the suggestion. 
It will be my pleasure to implement it :) 

Best,
Hugo