dqrobotics/matlab #85: [QUESTION] mismatch between the obtained target dummy in simulation (vrep) and matlab:

fahmizo, 2023-02-02:
the following code is aimed for controlling the pose of the end effector to the desired target pose of a dummy using the pseudoinverse controller:
both simulations run and at the same given instance, the manipulator in matlab displays error minimisation behavior till the target pose however, on vrep taken at the exact same instance the manipulator arm is quite far from the target pose desired.
[image]
[image]
and for the world frame, ive aligned both base and world reference frame to avoid further frame multiplication for testing purposes.
 the code used is also provided below:
[inverse controller for kuka code.txt](https://github.com/dqrobotics/matlab/files/10571342/inverse.controller.for.kuka.code.txt)

bvadorno, 2023-02-02:
Hi @fahmizo,

Thank you for posting your question and engaging with the DQ Robotics community. We have a strict policy in which anyone raising a question must accept the Code of Conduct automatically generated when you start an issue. Usually, when people fail to comply with this very simple request, we close the issue automatically. Nonetheless, since this is your first time, I'll guide you through the procedure.

1. First, include at the beginning of your question what you shouldn't have deleted:
> ### Acceptance of the Code of Conduct
> _Leave these in your question, read, and check them._ 
> - [ ] I understand that support for DQRobotics is given voluntarily. 
> - [ ] I understand that any possible response and its timeliness will be based on the relevance, accuracy, and politeness of my request and the following discussion.
> - [ ] If a DQRobotics member replies, I will let them know, politely, if their response solves my issue or not. 

2. The [MWE](https://stackoverflow.com/help/minimal-reproducible-example) should be placed directly in the issue you opened (i.e., this interface) using Markdown language, not in a separate file. You can find how to include code directly on the web interface [here](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code). An example of what we expect can be found [here](https://github.com/dqrobotics/matlab/issues/76).

3. Your code is currently irreproducible because you haven't provided the CoppeliaSim scene file you are using. You must do it (and being a CoppeliaSim file, not a source code, you can attach it.) Also, you haven't provided basic details: is the dynamics engine activated? Can you retrieve basic information from CoppeliaSim, such as the end-effector pose at different configurations? (This is something that you can easily compare with the MATLAB output.) What is the CoppeliaSim version you are using? What is your operating system? Also, without knowing if you can send and get information from CoppeliaSim **correctly**, it is pointless to discuss if the controller works or not.
 
After you solve all three aforementioned items, we can proceed.

I'm assigning @juanjqo to this issue, together with myself, and hopefully, we can help you with your problem.

Kind regards,
Bruno

juanjqo, 2023-02-06:
Hi @fahmizo, 

Welcome to the DQ Robotics community. Thank you for providing more information about your issue. I reproduced the behavior you mentioned, which is expected, according to your code and your scene. 

There are several contributing factors to your simulation not working.

### 1. Dynamics engine
The robot moves when you run the simulation from CoppeliaSim (without any script from Matlab).  This is because the dynamics engine is activated and there are low-level controllers acting on all joints. 

[image]

In your specific scene, you are using the joints in `dynamic mode` and `control position mode`. Therefore, there is a PID controller trying to minimize the error between the current angle and the target angle (This dynamic is handled by the physics engine, which in your scene is the Bullet engine). In those cases, you need to set the target angles by using `set_joint_target_positions(jointnames, qtarget)` instead of `set_joint_positions(jointnames, q)`.

[image]

If your goal is to use CoppeliaSim only as a visualization tool (no dynamics), you need to set the joints to `Kinematic Mode` and disable the dynamics of the robot links (uncheck the box `Body is dynamic` in Rigid Body Dynamic Properties).
Example:
[image]

[image]

### 2. Wrong joint names

It looks like you edited the joint names of the robot in your scene. However, the sixth joint is called `joint` in the scene and `joint6` in your code. 

```matlab
joints={'joint1','joint2','joint3','joint4','joint5','joint6','joint7'};
```
[image]

### 3. Wrong robot model 

According to your code, you are modeling the KUKA LBR4 manipulator. However, you are using on CoppeliaSim a different manipulator (KUKA LBR iiwa 14 R820).  This will lead to modeling errors. Furthermore, the KukaLwr4Robot() class does not provide the right kinematic model for the KUKA LBR4+ manipulator available on CoppeliaSim.

----------------------------------

# Working Example

[image]

Please find attached a minimal example using the KUKA LBR4+ manipulator with the joints in kinematic mode (no dynamics).

```matlab
clear all;
close all;
clc;
vi = DQ_VrepInterface();
vi.disconnect_all();
vi.connect('127.0.0.1',19997);
vi.start_simulation();

jointnames={'LBR4p_joint1','LBR4p_joint2',...
            'LBR4p_joint3','LBR4p_joint4',...
            'LBR4p_joint5','LBR4p_joint6',...
            'LBR4p_joint7'};

%---------------------Robot definition--------------------------%
LBR4p_DH_theta = [0, 0, 0, 0, 0, 0, 0];
LBR4p_DH_d = [0.200, 0, 0.4, 0, 0.39, 0, 0];
LBR4p_DH_a = [0, 0, 0, 0, 0, 0, 0];
LBR4p_DH_alpha = [pi/2, -pi/2, pi/2, -pi/2, pi/2, -pi/2, 0];
LBR4p_DH_type = double(repmat(DQ_JointType.REVOLUTE,1,7));
LBR4p_DH_matrix = [LBR4p_DH_theta;
                    LBR4p_DH_d;
                    LBR4p_DH_a;
                    LBR4p_DH_alpha
                    LBR4p_DH_type];
            
robot = DQ_SerialManipulatorDH(LBR4p_DH_matrix);
xbase = vi.get_object_pose('LBR4p_joint1');
robot.set_reference_frame(xbase);
robot.set_base_frame(xbase);
robot.set_effector(1+0.5*DQ.E*DQ.k*0.07);
%----------------------------------------------------------------

%---------------------Controller definition----------------------%
controller = DQ_PseudoinverseController(robot);
controller.set_gain(0.5);
controller.set_damping(0.01);
controller.set_control_objective(ControlObjective.Translation)
controller.set_stability_threshold(0.00001);
%----------------------------------------------------------------

q = vi.get_joint_positions(jointnames);
T = 0.01;
while ~controller.system_reached_stable_region()
    xd = vi.get_object_pose('target');
    u = controller.compute_setpoint_control_signal(q, vec4(translation(xd)));
    q = q + T*u;
    vi.set_joint_positions(jointnames, q);
    vi.set_object_pose('effector', robot.fkm(q))
    vi.set_object_pose('base', xbase);
end

vi.stop_simulation();
vi.disconnect();
```

[kinematicMode_noDynamics.zip](https://github.com/dqrobotics/matlab/files/10660379/kinematicMode_noDynamics.zip)

fahmizo, 2023-02-06:
hello, 
I hope you are okay and thank you for the welcoming.
your comments have been very constructive and enlightening, you've helped me point out other functions  for my specific model as well as point out what is wrong with my old code.
I really appreciate the detailed feedback,
thank you.

juanjqo, 2023-02-06:
@fahmizo I'm glad to help you. If you have any questions, please feel free to ask.

bvadorno, 2023-04-01:
@juanjqo, your minimal examples are awesome. I'm going to close this issue because it's been solved, but we should create a zero to hero tutorial using CoppeliaSim as well. Do you want to lead this initiative together with @ffasilva?  

All the best,
Bruno

closed 2023-04-01 by bvadorno

juanjqo, 2023-04-02:
Hi @bvadorno, sure! That would be great! I'll work on it.

Best regards,

Juancho