dqrobotics/matlab #90: [ENHANCEMENT] Expose `Q4`

mmmarinho, 2023-03-01:
### Acceptance of the Code of Conduct
_Leave these in your question, read, and check them._ 
- [X] I understand that support for DQRobotics is given voluntarily. 
- [X] In understand that any possible response and its timeliness will be based on the relevance, accuracy, and politeness of my request and the following discussion.
- [X] If a DQRobotics member replies, I will let them know, politely, if their response solves my issue or not. 

### Question

Hello @juanjqo,

Related to the matter: https://github.com/dqrobotics/cpp/issues/55

I've talked to @bvadorno today and we agreed that the `Q4` method should be exposed in MATLAB before we proceed with everything else.

Could you help us with that?
- Create the `Q4.m` file with the method
- Add the check for unit norm now that it will be a standalone method.

Thanks,
Murilo

juanjqo, 2023-03-01:
@mmmarinho sure thing! I would be happy to help!

bvadorno, 2023-04-04:
Fixed in #91.

closed 2023-04-04 by bvadorno