dqrobotics/matlab #92: [REQUEST]  Add coppeliasim_synchronization_mode methods to DQ_Interface

BomingHu123, 2023-03-10:
***
Acceptance of the Code of Conduct
Leave these in your bug report, read, and check them.

 I understand that support for DQRobotics is given voluntarily.
 I understand that any possible response and its timeliness will be based on the relevance, accuracy, and politeness of my request and the following discussion.
 If a DQRobotics member replies, I will let them know, politely, if their response solves my issue or not.
@dqrobotics/developers
***
**Problem describption**
Hi, I currently need to estimate robot velocity using robot pose information obtained from Coppeliasim. But I found that if the data transmission in synchronous mode of coppeliasim is not used, there will be a delay in data transmission, resulting in a large difference between the estimated speed and the actual data, shown in below.
[image]

**To Reproduce**
For example, this code creates a simple example to reproduce the problem.
```
function vrep_scene_dqcontroller_demo(simulation_parameters)

include_namespace_dq;

%% Simulation convenience flags
SHOW_FRAMES = simulation_parameters.show_frames;

%% Initialize V-REP interface
vi = DQ_VrepInterface;
vi.disconnect_all(); % For testing, can be removed for release
vi.connect('127.0.0.1',19997);
vi.start_simulation();

%% Initialize VREP Robots
youbot_vreprobot = YouBotVrepRobot('youBot',vi);

%% Load DQ Robotics kinematics
youbot = youbot_vreprobot.kinematics();

%% Initialize controllers
solver = DQ_QuadprogSolver;
youbot_controller = DQ_ClassicQPController(youbot,solver);
youbot_controller.set_control_objective(ControlObjective.Pose);
youbot_controller.set_gain(10);
youbot_controller.set_damping(0.01);

sampling_time = 0.01; % V-REP's sampling time is 50 ms.
total_time = simulation_parameters.total_time; % Total time, in seconds.

%% Initalize arrays to store all signals resulting from the simulation
max_iterations = round(total_time/sampling_time);
youbot_tracking_error_norm = zeros(1,max_iterations);
youbot_control_inputs = zeros(youbot.get_dim_configuration_space(),...
    max_iterations);
youbot_q_vector = zeros(youbot.get_dim_configuration_space(), ...
    max_iterations);
corin_angular_velocity_vrep = zeros(3,max_iterations);
corin_angular_velocity_matlab = zeros(3,max_iterations);

current_pose =  vi.get_object_pose('/youBot/');
last_pose = vi.get_object_pose('/youBot/');

%% Control loop
for t=0:sampling_time:total_time
    last_pose = current_pose;
    youBot_q = youbot_vreprobot.get_q_from_vrep_wheel();
    current_pose =  vi.get_object_pose( '/youBot/');
    [~, base_angular_velocity] = compute_estimated_velocity(current_pose,last_pose,sampling_time);
    [~,angular_w_vrep] = vi.get_body_velocity_from_vrep( '/youBot/');
        corin_angular_velocity_vrep(:,ii) = angular_w_vrep;
    corin_angular_velocity_matlab(:,ii) = base_angular_velocity;
    youBot_u = [10,-10,-10,10];
    youBot_q = youBot_q+youBot_u'*sampling_time;
    youbot_vreprobot.send_q_to_vrep_wheel(youBot_q);
    ii = ii+1;
end
T1 = 0:sampling_time:total_time;
figure(1)
plot(T,corin_angular_velocity_vrep(1,:))
hold on;
plot(T,corin_angular_velocity_vrep(2,:))
hold on;
plot(T,corin_angular_velocity_vrep(3,:))
hold on;
plot(T,corin_angular_velocity_matlab(1,:),':')
hold on;
plot(T,corin_angular_velocity_matlab(2,:),':')
hold on;
plot(T,corin_angular_velocity_matlab(3,:),':')
hold on;
title('velocity')
xlabel('Time/s') 
ylabel('velocity from vrep and matlab') 
legend('angular velocity from vrep rx','angular velocity from vrep ry','angular velocity from vrep rz',...
    'angular velocity from matlab rx','angular velocity from matlab ry','angular velocity from matlab rz')

%% End V-REP
vi.stop_simulation();
vi.disconnect();
end

function [estimated_derivative, angular_velocity] = compute_estimated_velocity(current_pose,last_pose,sampling_time)
x_trans =  last_pose' * current_pose;
kxi = 2*log(x_trans)/ sampling_time;
angular_velocity_DQ = kxi.P;
angular_velocity = vec3(angular_velocity_DQ);
estimated_derivative=0.5 * kxi * current_pose;
end
```
In addition, I added several functions in YouBotVrep without authorization to realize the function of geting and sending the target joint value to the wheels of youBot.
```
        function q = get_q_from_vrep_wheel(obj)
            base_arm_q = obj.vrep_interface.get_joint_positions(obj.wheel_names);
            q = base_arm_q;
        end

        function send_q_to_vrep_wheel(obj,q)
            obj.vrep_interface.set_joint_target_positions(obj.wheel_names,q(1:4));
        end
```
Besides, in order to gain the velocity of the robot from coppeliasim using remApi, I also add a function in DQ_Interface 
```
        function [v,w] = get_body_velocity_from_vrep(obj,handle)
            [returnCode,v,w]= obj.vrep.simxGetObjectVelocity(obj.clientID, obj.handle_from_string_or_handle(handle), obj.OP_ONESHOT);
        end 
```
Please check the attachment to get the ttt file of the coppeliasim model for reproduction. 
[simple_test.zip](https://github.com/dqrobotics/matlab/files/10944478/simple_test.zip)

**Modification**
I modified the DQ_Interface class without authorization, and added two functions, synchronous and synchronous_trigger, to realize the synchronous mode of coppeliasim and matlab. code show as below
```
        function synchronous(obj)
            obj.vrep.simxSynchronous(obj.clientID,true);
        end
        
        function synchronous_trigger(obj)
            obj.vrep.simxSynchronousTrigger(obj.clientID);
        end
```
In the end, I achieved the result I wanted. The robot speed read from vrep basically fits the speed calculated by numerical approximation using pose in matlab. The effect is as shown in the figure.
[image]

**Request**
I've discussed this with @bvadorno, and I hope to add synchronous mode functionality to the currently DQ_Interface class.

**Environment**

- Windows 11
- dqrobotics version(current version of the master branch) 
- MATLAB 2018a

mmmarinho, 2023-03-16:
@BomingHu123 

Thanks for the report.

@bvadorno 

Indeed, I think this functionality is not ready in MATLAB, but it has been available in CPP for a while.
- https://github.com/dqrobotics/cpp-interface-vrep/blob/894d97605f5fcb2b94def7fc4b35fefcc24d064a/include/dqrobotics/interfaces/vrep/DQ_VrepInterface.h#L87

For MATLAB we have to move on with this issue (I think?) being spearheaded by @juanjqo (right?)
- https://github.com/dqrobotics/matlab/pull/67

If this is holding you back we can split that pull request into smaller ones.

juanjqo, 2023-05-10:
@dqrobotics/developers 

Hi @BomingHu123, 

The functionalities you requested were [added](https://github.com/dqrobotics/matlab/pull/97) recently in the master branch. 

Best regards, 

Juan

mmmarinho, 2023-05-16:
Fixed in #97

closed 2023-05-16 by mmmarinho